FactbirdHQ/atat #2: TODO: Investigate supporting futures and async/await

MathiasKoch, 2019-12-18:


FrozenDroid, 2022-02-04:
Bumping this issue to let people know that I am working on this. At our organization, we're using Embassy.
As I start work on this, I will update this issue with things that are left to explore and decide.

FrozenDroid, 2022-02-10:
I had a chat with @MathiasKoch yesterday to investigate what needs to happen.
One main thing is that we need a way to propagate events so that we can await them where necessary. https://github.com/jamesmunns/bbqueue/pull/86 looks quite promising but has stalled. Maybe we can pick this up and finish it. 
https://github.com/burrbull/fugit-timer will also need async traits. This shouldn't be too much work.

As for async in traits, I'm currently using https://github.com/dtolnay/async-trait. This is less than ideal due to the use of `Box` and thus alloc.
We might want to write our own async traits using GATs for now, or look into https://github.com/4lDO2/real-async-trait-rs (though I remember having some issues with this...)

MathiasKoch, 2022-02-23:
Thats a shame. Can i ask what you chose as alternative way?

rmja, 2023-02-17:
@MathiasKoch I made the client async over here:
https://github.com/rmja/chip-drivers/tree/master/cellular/simcom/src/atat_async

It would be really nice if that work could be contributed back to here. 
In addition to the async work, it also has some minor adjustments compared to the current upstream client.

Please take a look and let me know what you think.

MathiasKoch, 2023-02-20:
@rmja That looks amazing!

I would love to get you changes upstream! 
Most of the changes looks great! 

Could you make a PR, moving the blocking client into a blocking module, and adding the async one in an asynch module (behind an async feature-flag)?
It would also be great if you could add the same changes to the blocking client (decoding the framing header), perhaps even make it use the blocking version of embedded_io? 

I am a bit split on the change to embedded-time, even though I think the removal of all the const generics is a nice improvement. Do you have a reference implementation of `Clock` somewhere?

closed 2023-03-25 by MathiasKoch