FactbirdHQ/atat #6: Future Plans about AT command generation

tl8roy, 2020-01-10:
I am currently writing a similar crate to interface with AT modules. I see that you have actually got something talking to a module which is far more progressed than I am.

Where my implementation differs however is that I am using traits for both creating and parsing the commands as well as sending and receiving data.

My main question is if you had any thoughts on how you want to generate the AT commands? I have created a macro that will expand an enum out into a send command, state and received data enums. These are defined by:

```
#[atat("\n")]
        enum TelitCommand {
            AT (None,(u8,u16), ()),
            AT_ (None,(u8,u16), (i64)),
            AT__ (None,(&'b[u8]), (u32)),
        }
```
This expands out to being able to send AT=u8,u16 and returns an OK(()), AT?u8,u16 and returns an i64 and AT?=[u8] and returns a u32.

The state at the moment is that the crate can create the command and send it to a generic buffer. I have started the parse side and that will take a generic buffer and return the parsed data.

I also plan to have a straight command pass through for unusual commands.

MathiasKoch, 2020-01-10:
Cool.

This crate currently works really great for AT commands (request, response and unsolicited responses), but lacks a bit when you want to use it in a socket mode, or pure data mode.

This is what i am working on currently though.

I am still refining the traits of this crate a bit, to be able to do all that i want it to.

I just pushed my latest changes, for you to see. You can have a look in tests.rs for a simple usage example with just AT Commands..

Further more this type of interface allows for extended usages like:

```rust
#[derive(Debug, Clone)]
pub enum ResponseType {
    SingleSolicited(Response),
    // MultiSolicited(Vec<Response, heapless::consts::U10>),
    Unsolicited(UnsolicitedResponse),
    ExtendedData(Packet),
    None,
}

#[derive(Debug, Clone)]
pub enum RequestType {
    Cmd(Command),
    Data(Vec<u8, consts::U1024>),
    ExtendedData(Packet),
}

impl ATRequestType for RequestType {
    type Command = Command;

    fn try_get_cmd(self) -> Option<Self::Command> {
        match self {
            RequestType::Cmd(c) => Some(c),
            RequestType::ExtendedData(_p) => {
                // Attempt to extract any AT Cmd
                None
            }
            _ => None
        }
    }

    fn get_bytes(&self) -> &str {
        match self {
            RequestType::Cmd(c) => c.get_cmd().as_str(),
            RequestType::ExtendedData(p) => {
                p.pack().as_str()
            }
            _ => ""
        }
    }
}

impl ATCommandInterface for Command {
    type Response = ResponseType;
    ...
}
```

This code currently does not compile though, as i am battling the borrowchecker regarding `ATRequestType`

I am working actively on this every day though, and if you think it looks good, I would be happy to work out some collaboration..

My hope is to make a crate that can be widely used for AT devices in the rust community, although there is still a lot of potential optimizations..

My goal is to have it fully no_std, and heapless (maybe an alloc feature gate eventually for Box optimizations).

MathiasKoch, 2020-01-10:
Also you can have a look at https://github.com/BlackbirdHQ/ublox-short-range-rs for a real life implementation.

It comes with a working linux example in the examples folder.

tl8roy, 2020-01-10:
From what I have seen, I think some of your ideas are better than mine.
I still think it should be possible to automatically generate 80% of the commands just by describing the command, inputs and output in a enum or struct. Having some specific types to deal with those odd cases is a good idea.

For me, the end goal is being able to send a UPD or TCP packet without having to worry about all the fiddly bits of AT modules.

The other use case is a lot of modules also have GPS or other location services that can be extracted. Making that simple would be nice.

FWIIW, here is the output of the macro. Might help with your borrowchecker issues.
```
# [derive (Debug, Clone, Copy)]
 enum TelitCommand
{ AT, AT_, AT__, _CGI, UBD,
  RawCommandPassthrough 
} 

use ufmt :: uwrite ; 

#[derive (Debug)] 
enum TelitCommandCreate < 'b >
{ AT (u8, u16), AT_ (u8, u16), AT__ (), 
RawCommandPassthrough (& 'b [u8]), }

impl < 'a > ATATCreate < 'a > for TelitCommandCreate < '_ >
{
    type ATATState = TelitCommand ; 
    type Buffer = & 'a mut Cursor < [u8] > ;
    fn create_command (& self, buffer_out : Self :: Buffer) -> Result <TelitCommand, ATATError >
    {
        match self
        {
            Self :: AT (element_0, element_1) =>
            {
                uwrite!(buffer_out, "AT+AT={}{}\n", element_0, element_1).map_err (| _e | ATATError :: CreateError) ? ; 
               Ok (TelitCommand :: AT)
            }, Self :: AT_ (element_0, element_1) =>
            {
                uwrite!(buffer_out, "AT+AT?{}{}\n", element_0, element_1) .map_err (| _e | ATATError :: CreateError) ? ; 
               Ok(TelitCommand :: AT_)
            }, Self :: AT__ () =>
            {
                uwrite!(buffer_out, "AT+AT=?\n",) . map_err(| _e | ATATError :: CreateError) ? ; 
                Ok(TelitCommand :: AT__)
            }, Self :: RawCommandPassthrough (data) =>
            {
                for i in data . iter ()
                {
                    uwrite ! (buffer_out, "{}", i) . map_err
                    (| _e | ATATError :: CreateError) ? ;
                } Ok (TelitCommand :: RawCommandPassthrough)
            }, _ => Err (ATATError :: UnknownCommandError)
        }
    }
} 
enum TelitCommandParse { AT (), AT_ (), AT__ (), }

 impl < 'a > ATATParse <'a > for TelitCommandParse
{
    type ATATState = TelitCommand ;
     type Buffer = & 'a mut Cursor < [u8] > ;
    fn parse_response (state : TelitCommand, buffer_in : Self :: Buffer) ->
    Result < Self, ATATError >
    {
        match state
        {
            TelitCommand :: AT => { Err (ATATError :: UnknownCommandError) },
            TelitCommand :: AT_ => { Err (ATATError :: UnknownCommandError) },
            TelitCommand :: AT__ =>
            { Err (ATATError :: UnknownCommandError) }, _ => Err
            (ATATError :: UnknownCommandError)
        }
    }
}
```

I'll post my code to github anyway and see if we can agree on a common direction.

tl8roy, 2020-01-18:
So my code is now available [here](https://github.com/tl8roy/atat). I think the best way for me to proceed is to implement generic traits that you can then call.

I think I will need to add markers to the enum to do various special things.

MathiasKoch, 2020-01-19:
That look great! I will try to have a thorough look, when I am back at the office Tuesday. 

I think we should be able to make a generic crate for every one to use.

PR #11 (2020-02-12): Refactor/derive
Rework entire command and response structure, along with corresponding traits.
Add derives for command and response traits, along with serializers and deserializers.
Add serde based serializer and deserializer for AT commands and responses.
Split TX and RX for much less copy work in the client.
Implement ufmt uDebug and uDisplay for all commands.
Rename crate from at-rs to atat.
    
Fixes #5
Fixes #8
Fixes #9
Fixes #6
closed 2020-02-12 by MathiasKoch