FactbirdHQ/atat #8: Crate name

dbrgn, 2020-01-16:
The crate name is a bit confusing since someone [is blocking the `at` name on crates.io](https://crates.io/crates/at). The fact that you're aliasing `at_rs as at` in the examples doesn't help either :slightly_smiling_face: 

What about renaming it to `atat` in reference to [AT-AT](https://starwars.fandom.com/wiki/All_Terrain_Armored_Transport)? Star wars references already [have some tradition in Rust crates](https://github.com/sfackler/r2d2). The name is short and still available on crates.io.

AT-AT could also be made into a nice logo :)

MathiasKoch, 2020-01-17:
I have no strong feelings regarding the naming of the crate, so i'm up for it.

You know, the hardest thing in software development is naming things ;)

Would you maybe open a PR changing stuff? Then i will see if i can work out how to change the crates.io.. Yank and publish on the new name?

tl8roy, 2020-01-17:
Probably a good thing I saw this as I named my project ATAT. I'll let you have it on crates.io and merge my work into yours

dbrgn, 2020-01-17:
> Probably a good thing I saw this as I named my project ATAT-

:joy: 

Another idea I had was "Attila", in case someone wants to name a crate after a gory warlord :slightly_smiling_face:

PR #11 (2020-02-12): Refactor/derive
Rework entire command and response structure, along with corresponding traits.
Add derives for command and response traits, along with serializers and deserializers.
Add serde based serializer and deserializer for AT commands and responses.
Split TX and RX for much less copy work in the client.
Implement ufmt uDebug and uDisplay for all commands.
Rename crate from at-rs to atat.
    
Fixes #5
Fixes #8
Fixes #9
Fixes #6
closed 2020-02-12 by MathiasKoch