FactbirdHQ/atat #9: RFC: Refactor command structure

MathiasKoch, 2020-01-24:
I am opening this issue as an invitation to discuss the way this crate expects implementors to structure AT commands.

## TLDR 
> Can we find a better structure for the commands/responses, that will still align with the objectives and thoughts of this crate, while making it easier to create/maintain/derive and/or will optimize better for memory/processing.
> **MUST be `no_std` and should be able to run without `alloc`**

## Current way
Just as a sum up, the current way expects commands to look something along the lines of and enum with commands as,

```
use core::fmt::Write;
use heapless::{ArrayLength, String, Vec};

use at::{utils, ATCommandInterface, ATRequestType, MaxCommandLen, MaxResponseLines};

#[derive(Debug, Clone, PartialEq)]
pub enum Command {
    AT,
    GetManufacturerId,
    GetModelId,
    GetFWVersion,
    GetSerialNum,
    GetId,
    SetEcho { enable: bool },
    GetEcho,
}

#[allow(dead_code)]
#[derive(Debug)]
pub enum Response {
    ManufacturerId {
        id: String<MaxCommandLen>,
    },
    ModelId {
        id: String<MaxCommandLen>,
    },
    FWVersion {
        version: String<MaxCommandLen>,
    },
    SerialNum {
        serial: String<MaxCommandLen>,
    },
    Id {
        id: String<MaxCommandLen>,
    },
    Echo {
        enable: bool,
    },
    None,
}

impl ATRequestType for Command {
    type Command = Command;

    fn try_get_cmd(self) -> Option<Self::Command> {
        Some(self)
    }

    fn get_bytes<N: ArrayLength<u8>>(&self) -> Vec<u8, N> {
        self.get_cmd().into_bytes()
    }
}

impl ATCommandInterface for Command {
    type Response = Response;

    fn get_cmd<N: ArrayLength<u8>>(&self) -> String<N> {
        let mut buffer = String::new();
        match self {
            Command::AT => String::from("AT"),
            Command::GetManufacturerId => String::from("AT+CGMI"),
            Command::GetModelId => String::from("AT+CGMM"),
            Command::GetFWVersion => String::from("AT+CGMR"),
            Command::GetSerialNum => String::from("AT+CGSN"),
            Command::GetId => String::from("ATI9"),
            Command::SetEcho { ref enable } => {
                write!(buffer, "ATE{}", *enable as u8).unwrap();
                buffer
            }
            Command::GetEcho => String::from("ATE?"),
        }
    }

    fn parse_resp(
        &self,
        response_lines: &Vec<String<MaxCommandLen>, MaxResponseLines>,
    ) -> Response {
        if response_lines.is_empty() {
            return Response::None;
        }
        let mut responses: Vec<Vec<&str, MaxResponseLines>, MaxResponseLines> =
            utils::split_parameterized_resp(response_lines);

        let response = responses.pop().unwrap();

        match *self {
            Command::AT => Response::None,
            Command::GetManufacturerId => Response::ManufacturerId {
                id: String::from(response[0]),
            },
            _ => Response::None,
        }
    }

    fn parse_unsolicited(_response_line: &str) -> Option<Response> {
        Some(Response::None)
    }
}

```

But this makes a couple of things very difficult:
1. It makes doing macro generation of AT commands very difficult, due to the big coupled enum.
2. It means a number of rather large match cases for `get_cmd` & `parse_resp`, and would require the same match cases in order to implement eg. `max_timeout_time`, `can_abort` etc.

## Proposed way

So, because of this my suggestion would be to restructure this into something along the lines of a `ATCommand` trait in this crate:

```
pub trait ATCommand {
    type Response;

    fn cmd<N: ArrayLength<u8>>(&self) -> String<N>;

    fn can_abort(&self) -> bool {
        false
    }

    fn max_timeout_ms(&self) -> u32 {
        1000
    }

    fn response<N: ArrayLength<u8>, L: ArrayLength<String<N>>>(&self, lines: Vec<String<N>, L>) -> Self::Response;
}
```

This would allow implementors to use it as:

```
mod commands {
    use super::*;

    /// 4.12 Card identification +CCID
    /// Returns the ICCID (Integrated Circuit Card ID) of the SIM-card. ICCID is a serial number identifying the SIM.
    pub struct GetCCID;

    impl ATCommand for GetCCID {
        type Response = Option<()>;

        fn cmd<N: ArrayLength<u8>>(&self) -> String<N> {
            String::from("AT+CCID")
        }

        fn response<N: ArrayLength<u8>, L: ArrayLength<String<N>>>(
            &self,
            lines: Vec<String<N>, L>,
        ) -> Self::Response {
            None
        }
    }

    /// 5.3 Set module functionality +CFUN
    pub struct SetModuleFunctionality {
        pub fun: Functionality,
        pub rst: Option<ResetMode>,
    }

    impl ATCommand for SetModuleFunctionality {
        type Response = Option<()>;

        fn cmd<N: ArrayLength<u8>>(&self) -> String<N> {
            let mut b = String::new();
            write!(b, "AT+CFUN={}", self.fun as u8).unwrap();
            if let Some(rst) = self.rst {
                write!(b, ",{}", rst as u8).unwrap();
            }
            b
        }

        fn response<N: ArrayLength<u8>, L: ArrayLength<String<N>>>(
            &self,
            lines: Vec<String<N>, L>,
        ) -> Self::Response {
            None
        }

        fn max_timeout_ms(&self) -> u32 {
            1_000_000
        }

        fn can_abort(&self) -> bool {
            true
        }
    }
}
```

Making it much easier to create macros/proc_macros to implement these in a derive crate, while giving back implementors better handling of parsing responses etc.
Also it might make it easier to add more features to the commands (I have added the `max_timeout_ms` and the `can_abort` here, both with a default implementation.

## Issues

The remaining issue with this, that i have a hard time working around, is the decoupling queue

```
static mut REQ_Q: Option<Queue<RequestType, consts::U5, u8>> = None;
static mut RES_Q: Option<Queue<Result<ResponseType, ATError>, consts::U5, u8>,
        > = None;
```

Ideally I would love it if we could do:

```
static mut REQ_Q: Option<Queue<impl ATCommand, consts::U5, u8>> = None;
static mut RES_Q: Option<Queue<Result<<impl ATCommand>::Response, ATError>, consts::U5, u8>,
        > = None;
```

Or something equivalent? So this is where i leave this to brighter minds than mine. Enlighten me with the options available in this awesome language!

eugene2k, 2020-01-24:
What you want is dynamic dispatch. `impl ATCommand` is for static dispatch, Box<dyn ATCommand> would be for dynamic dispatch. The current version of rust book doesn't seem to talk about trait objects, so you have to refer to the previous version: https://doc.rust-lang.org/1.30.0/book/first-edition/trait-objects.html

MathiasKoch, 2020-01-24:
@eugene2k That would be awesome! But in my case, it should be alloc free, and `Box<>` is not :(

eugene2k, 2020-01-24:
Heapless has a `Pool` structure which allocates a generic `Box`. Std's `Box` is not unique in being able to use trait objects.

eugene2k, 2020-01-24:
Actually, I think I might be wrong here because it doesn't look like Heapless's `Box` works with unsized data. So, what you can do is (depending on your use case) either store `&dyn ATCommand` objects in the queue or `*const dyn ATCommand` while storing the real objects in `Box`es.

Michael-F-Bryan, 2020-01-24:
These are just some thoughts I had while skimming through the proposed API, feel free to scrutinise or disregard them completely :slightly_smiling_face: 

You probably shouldn't add  things like `N: ArrayLength<u8>` parameter to the trait methods. This means the types are no longer object safe, which makes it a pain when you need to handle different types of commands. Even if you don't have `Box`, you could create commands on the stack (or as static variables) and store them in an array/`Vec` (e.g. `&[&Dyn ATCommand]`).

It also feels funny to be returning a `String` from `get_cmd()`, normally I'd expect a getter to return a reference. 

For that matter, why does a command expose it's internals in such a way? Although Rust isn't your stereotypical OO language, I think this interface could benefit from Martin Fowler's [Tell, Don't Ask](https://martinfowler.com/bliki/TellDontAsk.html).

Why does the response handle owned data (`Vec<String<T>, L>`) instead of borrowed data? If you pass a `&str` from some buffer the response is written to, the implementor can call `.lines()` on to extract the response lines and you won't be copying hundreds of bytes around (just because we're not using the heap, doesn't mean these helpless data structures are cheap!) just to read a response. 

Passing in `heapless` types as parameters doesn't really add much over normal slices unless you want the user to mutate things. Even when parsing data it's quite common to take a `&str` or `&[u8]` and return a tuple of `Response` and the number of bytes read (E.g. `fn parse(data: &str) -> Result<(R, usize), ParseError>`). That way the caller can remove the appropriate number of bytes from its buffer, and it removes a lot of the cognitive overhead associated with type-level integers and generics.

MathiasKoch, 2020-01-24:
This is great! Keep the feedback coming, and i will do my best to improve the crate based on it. I am still fairly new to rust, so all feedback is awesome!

MathiasKoch, 2020-01-28:
@Michael-F-Bryan 
Taking your points one at a time, how would i handle 
> It also feels funny to be returning a String from get_cmd(), normally I'd expect a getter to return a reference.

Correctly in rust? 

I assume what you mean by reference is that it should return borrowed data `&str`, which i initially tried, but couldn't make work due to my very limited experience with lifetimes.

How would i return a slice from a function where i would need to format the contents of the slice based on input arguments?
eg.

```
fn get_cmd(&self) -> &str {
    match self {
        Command::Example { a } => ** String slice containing 'a'**
    }
}
```

I feel like if i could work out stuff like that, i would probably be able to remove a lot of owned data (String, Vec) and also a lot of the `N: ArrayLength<u8>` parameters that follows from them.

tl8roy, 2020-01-28:
ufmt needs to write to a ```?Sized``` type, so that is probably the starting point.

You could pass a buffer through ```get_cmd``` and just return Ok(()). That is not that unusual in no_std.

I am fairly sure I can make my macro work with anything once I have the traits defined (and an example)

eugene2k, 2020-01-28:
>How would i return a slice from a function where i would need to format the contents of the slice based on input arguments?

Returning a &str is not a hard rule. If your function takes arguments and creates a string out of them, then unless you also store this string internally you should return the string. If the function stores the string internally, then you should consider having it return a reference to the stored string instead of a clone of the string. The same goes for slices.

PR #11 (2020-02-12): Refactor/derive
Rework entire command and response structure, along with corresponding traits.
Add derives for command and response traits, along with serializers and deserializers.
Add serde based serializer and deserializer for AT commands and responses.
Split TX and RX for much less copy work in the client.
Implement ufmt uDebug and uDisplay for all commands.
Rename crate from at-rs to atat.
    
Fixes #5
Fixes #8
Fixes #9
Fixes #6
closed 2020-02-12 by MathiasKoch