FactbirdHQ/atat #32: Handle non-standard result codes

dbrgn, 2020-03-15:
When connecting to an AP with the ESP8266 fails, it returns something along the lines of

```
WIFI DISCONNECT\r\n
WIFI CONNECTED\r\n
+CWJAP:1\r\n\r\n
FAIL\r\n
```

It seems that the `FAIL` here is used instead of `ERROR`. According to [the reference](https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf), that `1` is an error code (timeout) and `FAIL` indicates the end of the response.

[image]

The AT spec (ITU V.250) does not mention such a result code. Can we handle non-standard manufacturer-specific result codes somehow?

dbrgn, 2020-03-15:
Ah, I read a bit further (section 5.7.2), it appears that the `+CWJAP` extended result can also be a final result code:

[image]

In that case the `+CWJAP:1\r\n` probably terminates the response, and `FAIL` can be ignored.

dbrgn, 2020-03-15:
Here's another case: When sending data from the ESP8266, the response code is `SEND OK\r\n`.

```
Sending HTTP request…
[2020-03-15T22:58:57Z TRACE atat::ingress_manager] Digest / Idle / ""
[2020-03-15T22:58:57Z DEBUG atat::client] Sending command: "AT+CIPSEND=72\r\n"
[2020-03-15T22:58:57Z TRACE atat::ingress_manager] Receiving 24 bytes
[2020-03-15T22:58:57Z TRACE atat::ingress_manager] Digest / Idle / "AT+CIPSEND=72\r\r\n\r\nOK\r\n> "
[2020-03-15T22:58:57Z TRACE atat::ingress_manager] Switching to state ReceivingResponse
[2020-03-15T22:58:57Z TRACE atat::ingress_manager] Digest / ReceivingResponse / "OK\r\n> "
[2020-03-15T22:58:57Z DEBUG atat::ingress_manager] Received response: Ok("")
[2020-03-15T22:58:57Z TRACE atat::ingress_manager] Switching to state Idle
[2020-03-15T22:58:57Z DEBUG atat::client] Sending command: "GET /?format=text HTTP/1.1\r\nHost: api.ipify.org\r\nUser-Agent: ESP8266\r\n\r\n"
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Receiving 17 bytes
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Digest / Idle / "> \r\nRecv 72 bytes\r\n"
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Clearing buffer with invalid response (incomplete: false, buflen: 19)
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Digest / Idle / ""
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Receiving 11 bytes
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Digest / Idle / "SEND OK\r\n"
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Clearing buffer with invalid response (incomplete: false, buflen: 9)
[2020-03-15T22:58:58Z TRACE atat::ingress_manager] Digest / Idle / ""
```

MathiasKoch, 2020-03-16:
So i think this is kinda two separate issues.

First is "custom" error codes. I have an idea on how to handle this properly. I will draft an RFC of my thought on how to tackle this, during today. Then you guys can see if that also fits with the ESP way of doing stuff, otherwise we will alter it a bit to cover all cases.

Second one is "custom" response codes, eg `SEND OK`. I think we should be able to incoorporate this into the `AtatCmd` trait? The same way the `max_timeout_ms` and `can_abort` is there.. 

Something like:

```
pub trait AtatCmd {
    /// The max length of the command.
    ///
    /// Example: For the command "AT+RST" you would specify
    ///
    /// ```
    /// type CommandLen = heapless::consts::U6;
    /// ```
    type CommandLen: ArrayLength<u8>;

    /// The type of the response. Must implement the `AtatResp` trait.
    type Response: AtatResp;

    /// Return the command as a heapless `String`.
    fn as_string(&self) -> String<Self::CommandLen>;

    /// Parse the string response into a `Self::Response` instance.
    fn parse(&self, resp: &str) -> Result<Self::Response, Error>;

    /// Whether or not this command can be aborted.
    fn can_abort(&self) -> bool {
        false
    }

    /// The max timeout in milliseconds.
    fn max_timeout_ms(&self) -> u32 {
        1000
    }

    /// Response code of the command.
    fn response_code(&self) -> &str {
        "OK"
    }
}
```  

Would that work?

dbrgn, 2020-03-16:
> That way it would default to OK, but you can overwrite it on certain commands.

That would be nice! Maybe it should be a string slice so that multiple response codes could be returned?

```rust
    /// Response codes of the command.
    fn response_codes(&self) -> &[&str] {
        &["OK", "SEND OK"]
    }
```

MathiasKoch, 2020-03-16:
Does that make sense? 

Do you have any commands that can have multiple response codes indicating "success" ?

MathiasKoch, 2020-03-16:
Anyhow.. Then we just need to figure out how to check for them in the best way possible.. 
I'm a bit occupied today, so if your up for it, take a swing at it.. Otherwise i will try whenever i get time.

dbrgn, 2020-03-16:
> Does that make sense?
>
> Do you have any commands that can have multiple response codes indicating "success" ?

I'm not sure yet, but it would give us some flexibility. On the other hand, maybe that's premature optimization.

MathiasKoch, 2020-03-18:
After thinking a bit, I think this solution would also allow us catch the "hacks" i have now made for `"@"` and `">"`, as they would essentially just be

```
/// Response code of the command.
fn response_code(&self) -> &str {
    ">"
}
```

on the "Prepare" part of a 2-part command.

Only issue i see, is the same as the custom FSM.. Getting the data from `response_code()` in the client, to the ingress manager?

dbrgn, 2020-03-18:
> Only issue i see, is the same as the custom FSM.. Getting the data from response_code() in the client, to the ingress manager?

As written in the chat, this communication is quite hard since it's asynchronous, especially avoiding race conditions will be very hard this way. What about not configuring new response codes on a per-command base but for the entire driver?

The application could then register new response and error codes on the ingress manager by adding matchers. The matchers could be functions that are applied on the incoming data. If the matcher matches, the command function would get the success or error information plus the text in the buffer, for parsing / handling.

MathiasKoch, 2020-03-18:
Hmm.. That could possibly work.
I think it would give quite a bit more work for the implementors of the drivers, but as long as we have default matcher that covers 90% of all AT usage, and we keep that up to date as well, it would probably be a nice addition to larger AT based drivers like mine, to be able to add custom matchers with device specific logic.

I think it would at least be a good approach until async/await is ready for the masses.

dbrgn, 2020-03-18:
Yep, and often certain devices use the same non-standard response / pattern for multiple commands. That way you'd register / encode that behavior once and can then use it across multiple commands.

> I think it would at least be a good approach until async/await is ready for the masses.

I'm not sure if async/await would help us in this case, we'd still need information flow from the command to the ingress manager and synchronization between the two.

MathiasKoch, 2021-05-25:
I think this can be solved by writing a custom digester and/or URC matcher?

If you have a look at the `DefaultDigester` here: https://github.com/BlackbirdHQ/atat/blob/master/atat/src/digest.rs and the `DefaultUrcMatcher` here: https://github.com/BlackbirdHQ/atat/blob/master/atat/src/urc_matcher.rs

Please feel free to re-open if this is not the case!

closed 2021-05-25 by MathiasKoch