FactbirdHQ/atat #42: [RFC] Custom ResponseCodeMatcher & ErrorCodeMatcher

MathiasKoch, 2020-03-28:
This is a proposal to add optional custom matchers for response codes and error codes in the same way we currently support a custom matcher for URC's.

The general structure of both would follow the UrcMatcher perfectly.

In case of the ublox cellular family, they support 3 different verbosity levels, resulting in:
- **Default:** `ERROR\r\n`
- **Numeric:** `+CME ERROR: 5\r\n`
- **Verbose:** `+CME ERROR: "Operation not allowed"\r\n`

Currently this crate only supports handling the default case, which i think makes sense, as the two others are very device specific. But a custom ErrorMatcher would open up for much better error handling in the device driver crate.

In the same way the ESP family has commands that on success returns `SEND OK`, as described in #32 . A custom ResponseCodeMatcher would allow handling these device specific response codes as well, without bloating the Atat crate.

Also it would move response codes `"@"` and `">"` into a custom response code matcher.

Question is then, would it make sense to unite all 3 types of matchers under one trait:
```
pub enum MatcherResult<L: ArrayLength<u8>> {
    NotHandled,
    Incomplete,
    UrcComplete(String<L>),
    ResponseComplete(String<L>),
    ErrorComplete(String<L>)
}

pub trait Matcher {
    type MaxLen: ArrayLength<u8>;

    fn process(&mut self, buf: &mut String<consts::U256>) -> MatcherResult<Self::MaxLen>;
}
```

dbrgn, 2020-03-28:
I like the concept of a combined matcher result :slightly_smiling_face: However, that would mean that an URC matcher can return a response.

Instead, there should probably be a single "custom matcher" type that is called as soon as no default response type can be detected.

For easier structuring of user code, it's would be nice if multiple matchers could be specified, in a certain order. That way, the user can write a response matcher and an URC matcher in separate impls. The order of registration would then define the order these traits are being called. However, without boxing it's really hard to store multiple matchers of different types in a data structure... Therefore I'm not sure if that's possible.

MathiasKoch, 2020-03-30:
> However, that would mean that an URC matcher can return a response.

Good point!

> Instead, there should probably be a single "custom matcher" type that is called as soon as no default response type can be detected.

I like this idea, in that case we should probably have two matchers a `UrcMatcher` and a `ResponseMatcher`.

The `UrcMatcher` would be exactly the one you implemented, and the `ResponseMatcher` could be something along the lines of 
```
pub enum ResponseMatcherResult<L: ArrayLength<u8>> {
    NotHandled,
    Incomplete,
    ResponseComplete(String<L>),
    ErrorComplete(String<L>)
}

pub trait ResponseMatcher {
    type MaxLen: ArrayLength<u8>;

    fn process(&mut self, buf: &mut String<consts::U256>) -> ResponseMatcherResult<Self::MaxLen>;
}
```

That way the `ResponseMatcher` call belongs during the `ReceivingResponse` state, and `UrcMatcher` during `Idle` state.

How does that sound?

Another thing is, how does the matchers distinguish what part of the result is to be removed from `buf` (i assume thats the full `String` in `MatcherResult`?), and what part to "forward" to the client as part of the actual response?

dbrgn, 2020-03-30:
> How does that sound?

That makes sense!

> Another thing is, how does the matchers distinguish what part of the result is to be removed from buf (i assume thats the full String in MatcherResult?), and what part to "forward" to the client as part of the actual response?

Generally I'd say the complete matched data should be removed and forwarded. It's a matcher, not a handler. But maybe there could be use cases where some kind of "normalization" could take place. Probably depends on the device?

MathiasKoch, 2020-03-30:
> Generally I'd say the complete matched data should be removed and forwarded. It's a matcher, not a handler. But maybe there could be use cases where some kind of "normalization" could take place. Probably depends on the device?

I totally agree it should be removed, but i am not so sure on the forwarded part..

Take this exchange as example:
```
>> AT+CCLK?
<< AT+CCLK?
<< +CCLK: "14/07/01,15:00:00+01"
<< OK
```
Where `<<` is the received part.

Ideally i would like it to remove the full `AT+CCLK?\r\n+CCLK: "14/07/01,15:00:00+01"\r\nOK\r\n` but i would like to forward only `+CCLK: "14/07/01,15:00:00+01"` to the client to be parsed.

And in this verbose error example:
```
>> AT+CGATT=1
<< AT+CGATT=1
<< +CME ERROR: No network service
```

I would want to remove the full `AT+CGATT=1\r\n+CME ERROR: No network service\r\n` but forward something like `No network service` wrapped in some Error type?

This is also how the current "default" matcher works

dbrgn, 2020-03-30:
Yep, you're right, things like echoed data should be removed before passing it on. However, I'd keep a separation between "matching" and "parsing" responses. The matcher would extract `+CME ERROR: No network service` while a handler(?) would turn that into a structured type.

Some unstructured brainstorming:

I haven't done much thinking about the handler part so far. It would be nice if the user could provide an error enum that has a [`FromStr`](https://doc.rust-lang.org/nightly/core/str/trait.FromStr.html) implementation. In contrast to responses (where we use structs with an `AtatResp` impl), there will probably be a finite amount of small success / error types.

On the other hand, some response codes might be specific to the request. `SEND OK` might indicate success after sending data, but would be invalid after setting the IP address.

MathiasKoch, 2020-03-30:
> Yep, you're right, things like echoed data should be removed before passing it on. However, I'd keep a separation between "matching" and "parsing" responses. The matcher would extract +CME ERROR: No network service while a handler(?) would turn that into a structured type.

For sure! I guess we agree on the scope now, just need an actual format :)

> I haven't done much thinking about the handler part so far. It would be nice if the user could provide an error enum that has a FromStr implementation. In contrast to responses (where we use structs with an AtatResp impl), there will probably be a finite amount of small success / error types.

I think this would be a great way, if it could work!

bobmcwhirter, 2020-08-24:
I think this looks good. 

The ESP does signal "end of a blob of bytes" in a few different ways. 

The SEND OK is a good point.  Likewise the `> ` to proceed with data xmit, and of course the weird response+FAIL when joining an AP fails.

Being able to parameterize end-of-blob markers is good thing, I think.

dbrgn, 2020-08-25:
@bobmcwhirter (offtopic) if you're working on ESP integration, check out https://github.com/dbrgn/espresso where some work is already done!

MathiasKoch, 2021-05-25:
I think this issue can be solved by writing a custom Digester? 

If you have a look at the `DefaultDigester` here: https://github.com/BlackbirdHQ/atat/blob/master/atat/src/digest.rs