FactbirdHQ/atat #46: Fix doctests and examples and add them to CI

MathiasKoch, 2020-04-17:
Currently the CI will only run code tests 
`cargo test --tests --all-features`

But it should probably be changed to run 
`cargo test --all-features`

This requires that the all the documentation compiles, and that the examples compiles.

This would allow unit testing of small helper functions in doctests, eg.

```
    /// Returns `true` if `needle` is a suffix of the Vec.
    ///
    /// Always returns `true` if `needle` is an empty slice.
    ///
    /// # Examples
    ///
    /// ```
    /// use heapless::Vec;
    /// use heapless::consts::*;
    ///
    /// let mut v: Vec<_, U8> = Vec::new();
    /// v.push("a").unwrap();
    /// v.push("b").unwrap();
    ///
    /// assert!(v.ends_with(&[]));
    /// assert!(v.ends_with(&["b"]));
    /// assert!(v.ends_with(&["a", "b"]));
    /// assert!(!v.ends_with(&["a", "b", "c"]));
    /// assert!(!v.ends_with(&["b", "a"]));
    /// assert!(!v.ends_with(&["a"]));
    /// ```
    pub fn ends_with(&self, needle: &[T]) -> bool
    where
        T: PartialEq,
    {
        let (v, n) = (self.len(), needle.len());
        v >= n && needle == &self[v - n..]
    }
```

dbrgn, 2020-04-17:
Note that some code in doctests can be hidden from the docs: https://doc.rust-lang.org/rustdoc/documentation-tests.html Can be quite useful if the test requires setup code that you don't want to show in the docs.

MathiasKoch, 2020-04-17:
Ahh, thats a neat feature! Will remember that, when i come around to fixing this issue..

I think the examples are gonna be the worst one though.. I am not sure what is wrong, but it seems they are not catching the compile target? is `.cargo/config` unused for examples?

MathiasKoch, 2020-04-17:
Ohh nvm. The problem is of course that running tests, the target becomes my host, where i can't compile RTFM for.

Any way to exclude single examples from tests?

dbrgn, 2020-04-17:
You can run the tests with a `--target` parameter. That should work, right?

You could also create a `rtfm` Cargo feature that would be required for those tests, see https://github.com/stm32-rs/stm32l0xx-hal/blob/master/Cargo.toml#L273-L275.

MathiasKoch, 2020-04-17:
`--target thumbv7m-none-eabi` doesn't seem to work here.

The required-feature thing was new to me, thats a nice feature!

PR #54 (2020-07-24): Fix documentation tests & examples
- Fix documentation tests & examples. 
- Add docs to CI tests. 
- Add grcov codecoverage. 
- Add codecov.io (https://codecov.io/gh/BlackbirdHQ/atat) 

Fixes #46
closed 2020-07-24 by MathiasKoch