FactbirdHQ/atat #57: Change logging to be defmt based

MathiasKoch, 2020-08-24:
It should be possible to cut out all `core::fmt` bloat, for significant binary size wins, by switching out the `log` crate with `defmt` (https://github.com/knurling-rs/defmt)

**Pros:**
- Reduced binary size
- Faster logging
- Fits with `defmt` binaries

**Cons:**
- Not as flexible as `log` in general
- No logging on non-embedded targets

Perhaps it would be possible to make a wrapper that lets users choose between `log` and `defmt` at compile time?

dbrgn, 2020-08-24:
There's also [ufmt](https://docs.rs/ufmt/0.1.0/ufmt/), but I assume if defmt is newer, it must have some advantages.

MathiasKoch, 2020-08-24:
It has the obvious advantage of doing formatting on host-side, so it has a MUCH lower runtime overhead.

dbrgn, 2020-08-24:
Ahh, I should have looked at it more closely... That is a great approach and saves some of the headaches I've been having with logging :)

MathiasKoch, 2020-08-24:
Yup, it should actually allow logging stuff like incoming data in realtime, without much of a performance loss.

dbrgn, 2020-08-24:
Maybe we could provide wrappers for the macros that log to either defmt or ufmt, depending on cargo flags.

MathiasKoch, 2020-08-24:
That sounds like the correct approach

PR #58 (2020-08-28): Add support for defmt logging
This PR will add support for defmt (https://defmt.ferrous-systems.com/introduction.html) logging.

It introduces a couple of new feature flags, and changes the old logging flag:

 - `log-logging`: Disabled by default. Enable log statements on various log levels to aid debugging. Powered by `log`.
 - `defmt-default`: Disabled by default. Enable log statements at INFO, or TRACE, level and up, to aid debugging. Powered by `defmt`.
 - `defmt-trace`: Disabled by default. Enable log statements at TRACE level and up, to aid debugging. Powered by `defmt`.
 - `defmt-debug` [... cut]
closed 2020-08-28 by MathiasKoch