FactbirdHQ/atat #59: Docs: Does the driver! macro still exist?

mon, 2020-10-02:
Import fails and I can't seem to find it in the atat source - should I create a Client manually?

MathiasKoch, 2020-10-02:
Hi @mon 

Yes, sorry for the back and forth of the initialization. Yes, you have to crate the Client manually. You can find an example here: https://github.com/BlackbirdHQ/atat/blob/master/atat/examples/rtic.rs

I will update the docs right away. Apparently they slipped by my review before releasing.

mon, 2020-10-02:
Great! All going well I should have a working example for getting the IMEI off a sim card using the `linux_embedded_hal` early next week!

MathiasKoch, 2020-10-02:
Cool!

This might be of help for you: https://github.com/BlackbirdHQ/ublox-cellular-rs/tree/master/examples

commit 8ba4beff (2020-10-06): Correct docs about how to instantiate atat Client. Fixes #59
commit 85503ddb (2020-10-06): Correct docs about how to instantiate atat Client. Fixes #59
closed 2020-10-06 by MathiasKoch