FactbirdHQ/atat #76: Support multiplexing

MathiasKoch, 2021-02-04:
It would be awesome to eventually add support for the `3gpp ts 27.010 multiplexer protocol` described in https://www.3gpp.org/ftp/tsg_t/tsg_t/tsgt_04/docs/pdfs/TP-99119.pdf

Ublox has an implementation guide for it here: https://www.u-blox.com/en/docs/UBX-13001887

Furthermore a C implementation for reference can be found here: https://github.com/particle-iot/gsm0710muxer/tree/d7f7b75e8d48af760c4f585f222287eb9eff9422

This would probably be as a separate crate in this workspace, and might require extensive changes to the ATAT crate itself, which would be fine. 

Any contributions are welcomed!

MathiasKoch, 2021-03-09:
I think the way forward on implementing this, is to do it similar to the way Extended Data Mode is implemented in https://github.com/BlackbirdHQ/ublox-short-range-rs/blob/master/ublox-short-range/src/command/custom_digest.rs

ckrenslehner, 2025-03-31:
@MathiasKoch are there still plans to add the multiplexing support to this crate?

In the `ublox-cellular` crate this embassy fork is utilized for parsing:
`embassy-at-cmux = { git = "https://github.com/MathiasKoch/embassy", rev = "66d78d425" }`

So it seems you have already done all the heavy lifting. 🚀

MathiasKoch, 2025-03-31:
It is still a bit undecided how the entire multiplexing stuff lands eventually, and whether it will be part of ATAT, or not.

For now, I have the PR you have referenced yourself, and we have added multiplexing to our ublox cellular crate next to ATAT. It's not perfect, but on the other hand, I am still not 100% convinced it can be generalized enough to make it part of ATAT without influencing non-multiplexed users a lot..

ckrenslehner, 2025-03-31:
Thanks. I think it is fine to depend on a separate crate for multiplexing. 👍