FactbirdHQ/atat #94: ESP32 AT-Echo derive not working

Heusini, 2021-05-26:
The ESP32 returns the sent request as an echo thus i cannot simply derive the AtatResp but have to manually parse the response.

Here is an example.
This is the request i send:
```rust
#[at_cmd("+CWJAP", WifiConnectResponse)]
pub struct WifiConnect {
  #[at_arg(position = 0)]
  pub ssid: String<64>,
  #[at_arg(position = 1)]
  pub pw: String<128>,
}
```
This is the raw response as returned from the esp32:
```at
AT+CWJAP="ssid","password"
WIFI CONNECTED
WIFI GOT IP

OK
```

It would be nice if the above response could be parsed with the following AtatResp derive.

```rust
#[derive(Clone, Debug, AtatResp)]
pub struct WifiConnectResponse {
  #[at_arg(position = 0)]
  pub connected: String<32>,
  #[at_arg(position = 1)]
  pub got_ip: String<32>,
}
```

MathiasKoch, 2021-05-26:
The problem is not actually that the device returns the echo, this is actually a requirement currently due to #82 .

The problem in this case, is that the ATAT parser expects parameters of the response to be separated by a `,`, where in this case they are separated by `\r\n`

I will have a look when i get time, to see if it would be possible to extend the derive capabilities with a separator option.

Thanks for reporting!

Heusini, 2021-05-26:
Shouldn't this parse?
This is the full raw response:
```
AT+CIPDNS?
+CIPDNS:1,"42.2.129.13","8.8.8.8"

OK
```
This is the buffer after the digest:
```
+CIPDNS?
+CIPDNS:1,"42.2.129.13","8.8.8.8"
```

This is the struct i use to parse:
```rust
#[derive(Clone, Debug, PartialEq, AtatResp)]
pub struct DNSResponse {
    pub dns: String<64>,
    pub dns_1: String<64>,
    pub dns_2: String<64>,
}
```
shouldn't the DNSResponse look like this:
```
dns = +CIPDNS?\r\n+CIPDNS:1
dns_1 = 42.2.129.13
dns_2 = 8.8.8.8
```

MathiasKoch, 2021-05-27:
Hmm, not quite.

The parsed values will only be the parameters after the `:`, meaning that what you would probably want for this case would be:

````rust
#[derive(Clone, Debug, PartialEq, AtatResp)]
pub struct DNSResponse {
    pub some_flag: u8,
    pub dns_1: String<64>,
    pub dns_2: String<64>,
}
```

Where `some_flag` is the first parameter (1), that i have no idea what is without context :) 

That should work.. If you want to improve it a little further, you could use `no_std_net::IpAddr` for the IP's (or no_std_net::Ipv4Addr if only v4 is supported)

Just make sure to enable the `serde` feature of `no_std_net`

closed 2021-11-08 by MathiasKoch