FactbirdHQ/atat #102: derive(AtatCmd) cannot parse numeric string

apeng2012, 2021-09-03:
my at commands:

```
AT+CGSN
816363036697552

OK
```

without atat_derive no problem.
``` rust
pub struct RequestSN;

impl atat::AtatCmd<9> for RequestSN {
    type Response = Str64Response;
    type Error = GenericError;

    fn as_bytes(&self) -> Vec<u8, 9> {
        Vec::from_slice(b"AT+CGSN\r\n").unwrap()
    }

    fn parse(&self, resp: Result<&[u8], &InternalError>) -> Result<Self::Response, Error<Self::Error>> {
        Ok(Str64Response {
            s: String::from(core::str::from_utf8(resp.unwrap()).unwrap())
        })
    }
}
```

But using atat_derive does not work
``` rust
#[derive(Clone, AtatCmd)]
#[at_cmd("+CGSN", Str64Response, timeout_ms = 1000)]
pub struct RequestSN;
```

``` rust
#[derive(Clone, AtatResp)]
pub struct Str64Response {
    #[at_arg(position = 0)]
    pub s: String<64>,
}
```

MathiasKoch, 2021-09-03:
I see a few issues in this, but i think the main reason this is not working for you is that by definition in AT commands, a string is quoted, meaning the response parameter you get back here is not a string.

You have two options for parsing this correctly. 

Either you change it to be a suiting integer, probably u128 in this case.
Or you could use `heapless_bytes` to get it as a byte array
```
#[derive(Clone, AtatResp)]
pub struct Str64Response {
    #[at_arg(position = 0)]
    pub s: heapless_bytes::Bytes<64>,
}
``` 

Both of those should work directly for you.

apeng2012, 2021-09-06:
Both methods can be used.
Thank you very much.

Because I have a return in this format “170.100.00.01.06”, so I chose the latter.

closed 2021-09-06 by apeng2012

MathiasKoch, 2021-09-06:
Sounds good, glad you mae it work :+1: