FactbirdHQ/atat #149: ATI response cannot be parsed as String, but can as Bytes

ijager, 2023-04-29:
I noticed the following when trying out `atat`. When defining ATI response using `heapless::Bytes` it parses correctly.

```rust
#[derive(Debug, Clone, AtatResp)]
pub struct ATIResponse
{
    pub info: Bytes<64>,
}
```

The response:

```rust
ATIResponse { info: b'Quectel\r\nEC25\r\nRevision: EC25EFAR06A09M4G' }
```

But when using `heapless::String` it results in `Error::Parse`.

```rust
#[derive(Debug, Clone, AtatResp)]
pub struct ATIResponse
{
    pub info: String<64>,
}
```

Perhaps still the same issue as in #86.

MathiasKoch, 2023-05-01:
Hmm. Without any more context, it's a bit hard to tell why, but the main difference between bytes & string, is that a string reply must be quoted explicitly (So you cannot just parse a bytes response as a string response currently)

ijager, 2023-05-01:
Ah so I guess this is intended behaviour then. So only when the modem returns explicit `"...."` including the quotes we can use `String`

MathiasKoch, 2023-05-01:
> Ah so I guess this is intended behaviour then. So only when the modem returns explicit `"...."` including the quotes we can use `String`

Currently, this is the behavior, yes. 

I would have no objections against adding some sort of derive attribute option to the string implementation, that allows skipping the quotes in either end? I currently have no use cases for this myself, so I won't be adding it, but you are more than welcome to open a PR with it? :smiley:

jamwaffles, 2023-05-01:
I ran into this problem at work and came up with the following solution. I'm not particularly proud of it but it's an ok stopgap until something nicer comes along :)

```rust
/// A helper to deserialize unquoted strings in AT responses.
struct UnquotedStringVisitor<const N: usize>;

impl<'de, const N: usize> Visitor<'de> for UnquotedStringVisitor<N> {
    type Value = heapless::String<N>;

    fn expecting(&self, formatter: &mut fmt::Formatter) -> fmt::Result {
        formatter.write_str("an unquoted string")
    }

    fn visit_bytes<E>(self, v: &[u8]) -> Result<Self::Value, E>
    where
        E: de::Error,
    {
        let s = std::str::from_utf8(v).map_err(serde::de::Error::custom)?;

        if s.len() > N {
            Err(serde::de::Error::custom("Source string too long"))
        } else {
            Ok(s.into())
        }
    }
}

#[derive(Debug, Clone, PartialEq)]
pub struct UnquotedHeaplessString<const N: usize>(pub heapless::String<N>);

impl<'de, const N: usize> Deserialize<'de> for UnquotedHeaplessString<N> {
    fn deserialize<D>(deserializer: D) -> Result<Self, D::Error>
    where
        D: serde::Deserializer<'de>,
    {
        let out = deserializer.deserialize_bytes(UnquotedStringVisitor)?;

        Ok(Self(out))
    }
}

impl<const N: usize> From<&str> for UnquotedHeaplessString<N> {
    fn from(s: &str) -> Self {
        Self(s.into())
    }
}

impl<const N: usize> fmt::Display for UnquotedHeaplessString<N> {
    fn fmt(&self, f: &mut fmt::Formatter<'_>) -> fmt::Result {
        f.write_str(&self.0)
    }
}
```

mschnell1, 2023-05-05:
Using MQTT on a Quectel,  I also see that I can't get the result of the messages, which feature two texts included in double quotes, when defining the elements in the OnReceive struct as heapless strings. 
-Michael

mschnell1, 2023-05-06:
@ijager 
BTW.: 
In fact I did get such a parser error, when Trying `ATI` but instead of trying out "Bytes" in a response struct, I avoided using the parser by not deriving the "send" - struct but implementing my own parser. 
IMHO this in fact makes sense for not "true" URC (but direct on send()) responses from the modem, as here you easily can handle the response in any way you like.

-Michael

mschnell1, 2023-05-06:
> I would have no objections against adding some sort of derive attribute option to the string implementation, that allows skipping the quotes in either end? I currently have no use cases for this myself, so I won't be adding it, but you are more than welcome to open a PR with it? smiley

Are there already any derive attributes that define the behavior of the parser differently from what the main derive macro extracts from the type of the struct elements ?

MathiasKoch, 2023-05-06:
I think you are mixing request/response and URCs here. 

For the request response it is all handled by AtatCmd and AtatResponse that can both be derived, URC parser has nothing to do with these at all

mschnell1, 2023-05-06:
I'm not in the office right now. I'll do more testing ASAP. (Hoping to be able to avoid the implementation of my own parser <which does work fpr now> for the response of communication as simple as `ATI` with non quoted strings <by means of  `#[at_arg(position = 0)]` or similar>. And of course to get a notification in the user code from an urc with quoted strings such as `+QMTRECV`.

ijager, 2023-05-07:
@mschnell1 I am also implementing Quectel MQTT commands for the EC25. Just managed to connect to a broker. Next step I think is also those URC's and some state machine to progress through the necessary commands.

I did notice that other commands such as `ManufacturerId` (`AT+CGMI`) do get parsed fine into a `heapless::String` even though those responses are _not_ quoted as far as I know (at least not from my Quectel EC25). So far the only difference I can see is that the `ATI` response is multi line instead of a single line ending in `\r\n`. Which I guess makes sense that it would trip up the default String parsing, because I think it is usually only reading until the first `\r\n`.

@MathiasKoch I am not particularly interested in the `ATI` response, so for me it is enough to know that `Bytes` + manual parsing is the way to go for those cases. If you want you can close this issue.

mschnell1, 2023-05-07:
> as far as I know
You can use the log feature of ATAT to see the complete dialog with the modem.

ijager, 2023-05-08:
Yes when I turn on logging for atat it shows quotes, but when I sniff the communication with a serial - usb converter there are no quotes. So maybe the quotes are added somewhere in atat?

I will try to implement subscribing today.

mschnell1, 2023-05-08:
Edit: 

by some magic I now get a correct +QMTRECV notification by ATAT

MathiasKoch, 2023-05-08:
Well, if you are receiving URC's that are not defined in the `Urc` enum, it will kinda end up as undefined behaviour, as it will become part of the next attempt to parse a known URC or response, with varying success.

ijager, 2023-05-08:
@mschnell1 I got it working now, the `QMTRECV` parses okay. But indeed you have to define and decorate the URC enum and define the struct properly so that it matches your settings. E.g. whether it returns the length as well.

This is my Urc enum so far:

```rust
#[derive(Clone, AtatUrc, Debug)]
pub enum Urc {
    #[at_urc("+QMTOPEN")]
    MqttOpen(commands::mqtt::urc::MqttOpen),
    #[at_urc("+QMTCONN")]
    MqttConnect(commands::mqtt::urc::MqttConnect),
    #[at_urc("+QMTSUB")]
    MqttSubscribe(commands::mqtt::urc::MQTTSubscribe),
    #[at_urc("+QMTRECV")]
    MQTTReceive(commands::mqtt::urc::MQTTReceive)
}

// and the urc structs

#[derive(Clone, Debug, AtatResp)]
pub struct MqttOpen {
    pub client_idx: u8,
    pub result: u8,
}

#[derive(Clone, Debug, AtatResp)]
pub struct MqttConnect {
    pub client_idx: u8,
    pub result: u8,
    pub ret_code: u8,
}

#[derive(Clone, Debug, AtatResp)]
pub struct MQTTSubscribe {
    pub client_idx: u8,
    pub msg_id: u8,
    pub result: u8,
    pub value: u8,
}

#[derive(Clone, Debug, AtatResp)]
pub struct MQTTReceive {
    pub client_idx: u8,
    pub msg_id: u8,
    pub topic: String<32>,
    pub len: u8,
    pub payload: String<32>
}

```

Here is my full log output in case it helps:

```
14:54:35.576 DEBUG - [EC25] state: Init
14:54:35.680 DEBUG - send next cmd EchoOn(EchoOn)
14:54:35.680 DEBUG - Sending command: ""ATE1\r\n""
14:54:35.783 DEBUG - Received OK
14:54:35.864 DEBUG - send next cmd GetIdentification(IdentificationInformation)
14:54:35.864 DEBUG - Sending command: ""ATI\r\n""
14:54:35.969 DEBUG - Received response: ""Quectel\r\nEC25\r\nRevision: EC25EFAR06A09M4G""
14:54:36.071 DEBUG - send next cmd GetManufacturerId(GetManufacturerId)
14:54:36.071 DEBUG - Sending command: ""AT+CGMI\r\n""
14:54:36.172 DEBUG - Received response: ""Quectel""
14:54:36.172 DEBUG - Manufacturer: ManufacturerId { id: "Quectel" }
14:54:36.278 DEBUG - send next cmd GetModelId(GetModelId)
14:54:36.278 DEBUG - Sending command: ""AT+CGMM\r\n""
14:54:36.384 DEBUG - Received response: ""EC25""
14:54:36.384 DEBUG - Model: ModelId { id: "EC25" }
14:54:36.460 DEBUG - send next cmd GetSoftwareVersion(GetSoftwareVersion)
14:54:36.460 DEBUG - Sending command: ""AT+CGMR\r\n""
14:54:36.565 DEBUG - Received response: ""EC25EFAR06A09M4G""
14:54:36.565 DEBUG - sw version: SoftwareVersion { id: "EC25EFAR06A09M4G" }
14:54:36.668 DEBUG - send next cmd SetPDPContextDefinition(SetPDPContextDefinition { cid: ContextId(1), pdp_type: "IP", apn: "portalmmm.nl" })
14:54:36.668 DEBUG - Sending command: ""AT+CGDCONT=1,\"IP\",\"portalmmm.nl\"\r\n""
14:54:36.773 DEBUG - Received OK
14:54:36.877 DEBUG - send next cmd ActivateContext(ActivateContext { cid: ContextId(1) })
14:54:36.877 DEBUG - Sending command: ""AT+QIACT=1\r\n""
14:54:36.980 DEBUG - Received OK
14:54:37.061 DEBUG - send next cmd MQTTConfigRecvMode(ConfigRecvMode { param: "recv/mode", client_idx: ContextId(1), msg_recv_mode: 0, msg_len_enable: Some(1) })
14:54:37.061 DEBUG - Sending command: ""AT+QMTCFG=\"recv/mode\",1,0,1\r\n""
14:54:37.162 DEBUG - Received OK
14:54:37.264 DEBUG - send next cmd MQTTOpenContext(Open { client_idx: ContextId(1), host_name: "mybroker.xyz", port: 1883 })
14:54:37.264 DEBUG - Sending command: ""AT+QMTOPEN=1,\"mybroker.xyz\",1883\r\n""
14:54:37.370 DEBUG - Received OK
14:54:37.473 DEBUG - send next cmd AT(AT)
14:54:37.473 DEBUG - Sending command: ""AT\r\n""
14:54:37.579 DEBUG - Received OK
14:54:37.659 DEBUG - urc: MqttOpen(MqttOpen { client_idx: 1, result: 0 })
14:54:37.659 DEBUG - [EC25] state: Connect
14:54:37.764 DEBUG - send next cmd MQTTConnect(Connect { client_idx: ContextId(1), mqtt_client_id: "Ingmar" })
14:54:37.764 DEBUG - Sending command: ""AT+QMTCONN=1,\"Ingmar\"\r\n""
14:54:37.870 DEBUG - Received OK
14:54:37.870 DEBUG - MQTT CONNECT: NoResponse
14:54:37.976 DEBUG - urc: MqttConnect(MqttConnect { client_idx: 1, result: 0, ret_code: 0 })
14:54:37.976 DEBUG - [EC25] state: Subscribe
14:54:38.077 DEBUG - send next cmd MQTTSubscribe(Subscribe { client_idx: ContextId(1), msg_id: 1, topic: "blink", qos: 0 })
14:54:38.077 DEBUG - Sending command: ""AT+QMTSUB=1,1,\"blink\",0\r\n""
14:54:38.155 DEBUG - Received OK
14:54:38.155 DEBUG - MQTT SUB: NoResponse
14:54:38.261 DEBUG - urc: MqttSubscribe(MQTTSubscribe { client_idx: 1, msg_id: 1, result: 0, value: 0 })
14:54:38.261 DEBUG - [EC25] state: Ready
14:54:42.555 DEBUG - urc: MQTTReceive(MQTTReceive { client_idx: 1, msg_id: 0, topic: "blink", len: 5, payload: "start" })
14:54:42.555 INFO - Received "start" on topic "blink"
14:55:03.898 DEBUG - urc: MQTTReceive(MQTTReceive { client_idx: 1, msg_id: 0, topic: "blink", len: 5, payload: "hello" })
14:55:03.898 INFO - Received "hello" on topic "blink"
```

And this is what I see on the uart lines:

```console
RDY
AT
OK
ATE1
OK
ATI
Quectel
EC25
Revision: EC25EFAR06A09M4G
OK
AT+CGMI
Quectel
OK
AT+CGMM
EC25
OK
AT+CGMR
EC25EFAR06A09M4G
OK
AT+CGDCONT=1,"IP","portalmmm.nl"
OK
AT+QIACT=1
OK
AT+QMTCFG="recv/mode",1,0,1
OK
AT+QMTOPEN=1,"mybroker.xyz",1883
OK
AT
OK
+QMTOPEN: 1,0
AT+QMTCONN=1,"Ingmar"
OK
+QMTCONN: 1,0,0
AT+QMTSUB=1,1,"blink",0
OK
+QMTSUB: 1,1,0,0
+QMTRECV: 1,0,"blink",5,"start"
+QMTRECV: 1,0,"blink",5,"hello"
```

mschnell1, 2023-05-09:
> Well, if you are receiving URC's that are not defined in the `Urc` enum, it will kinda end up as undefined behaviour, as it will become part of the next attempt to parse a known URC or response, with varying success.

Nasty, but supposedly unavoidable. 

This seems to mean that any possible response by the modem needs to be covered in the Urc enum....

MathiasKoch, 2023-05-09:
So far i haven't found a better approach, but to be fair **most** URC's are default disabled on most modems, and needs to be explicitly enabled. At least on the modems i have used?

mschnell1, 2023-05-09:
@ijager 
Seeminglöy you use a differet logger. in my Debug output I don't see the double double quotes. It looks like this: 
`
DEBUG atat::ingress               > Received response (51/51): "Quectel\r\nEC25\r\nRevision: EC25EFAR06A14M4G"
`
I Can't fmnd the type "`[Bytes](heapless::Bytes)`" you use (the ATAT create does not republish is while it dose provide `atat::heapless::String` )
So I tried to do 
```

#[derive(Clone, Debug, AtatResp)]
pub struct QueryModemType2Response {
    #[at_arg(position = 0)]
    pub received_modem_type: atat::heapless::Vec<u8, 300>, //String<0x100>,
}
```
but I still get the send() result:  `Err(Parse)` 

Any further idea ? 

Btw.: is it appropriate to do such "silly user" discussions in an "issue" thread ?

-Michael

mschnell1, 2023-05-09:
Sorry for continuing being a PITA.....
1) I don't find out anything about the ` #[at_arg(position = 0)]`  stuff. ATAT seems somehow to dorward this to some serdes functionality. Any starters on how to find any documentation on that ? 

2) Right now I use `AT+QMTPub` to bublish QMT topic content. That6 does work and easily is compatible with ATAT. 
But in fact that command ios nopt even documented with the Quectel modem I use. The documentation sttes AT+QMTPUBEX. I in fact would like to use that, anyway, as it can send binary data  instead of text only. 

With `AT+QMTEX`  the modem is expects a message length instead of the message as the 6th parameter. it then sends a `>`  and expects as many bytes as given as message length. After thet it sends `OK` .

Is such a procedure implemented ot anyhow supported in ATAT ? 

3) when transferring binary data with QMTT, Will ATAT be able to receive such  `+QMTRECV`  urc messages in the normal way ? 

Thanks for listening !
-Michael

MathiasKoch, 2023-05-09:
at_arg is documented as part of the relevant derive macro in atat_derive; https://github.com/BlackbirdHQ/atat/blob/master/atat_derive/src/lib.rs#L141-L165

MathiasKoch, 2023-05-09:
As for two this is supported and i use it multiple places, but for now it requires you to split it into two commands as

https://github.com/BlackbirdHQ/ublox-cellular-rs/blob/master/ublox-cellular/src/command/device_data_security/mod.rs#L79-L121

And then they can be used as:

https://github.com/BlackbirdHQ/ublox-cellular-rs/blob/master/ublox-cellular/src/services/data/ssl.rs#L55-L70

ijager, 2023-05-09:
> @ijager Seemingly you use a different logger. 

I am using `rtt-target` in combination with `log`.

> I Can't find the type `heapless::Bytes` you use 

It is actually a different crate: `heapless_bytes::Bytes`.

> Btw.: is it appropriate to do such "silly user" discussions in an "issue" thread ?

There is also the [matrix chat](https://matrix.to/#/!ocRyOwQJhEWrphujkM:matrix.org?via=chat.berline.rs&via=matrix.org)

mschnell1, 2023-05-10:
@MathiasKoch 
>As for question 3, i think that requires more knowledge about the modem than i have at present, and sounds less like an atat question?

As far as I understand,  the Modem can handle binary payload in such way: 

- configure the Modem to enable the optional length parameter in the `+QMTRCV `Urc
-  use `AT+QMTPUBEX `instead of the (not even documented but working `AT+QMTPUB `command to send the binary payload.
- When receiving the `+QMTRCV` Urc contains the topic in quotes, the length as digits, and then the payload in quotes 

Now the payload can contain any characters including quotes, comma, etc. Only by relying on the length parameter the receiver can know the end of the payload. The "URC parser" needs to  support this in some way.
-Michael

MathiasKoch, 2023-05-10:
Ahh, yeah that is a very common way of doing it in AT commands, but it is not currently supported in ATAT :/

Contributions are welcomed.

mschnell1, 2023-05-10:
@ijager 
> It is actually a different crate: heapless_bytes::Bytes.

Yep that works. 
Seemingly the Derive Macro recognizes `Bytes` but not `Vec<u8, 300>` .

mschnell1, 2023-05-11:
> Ahh, yeah that is a very common way of doing it in AT commands, but it is not currently supported in ATAT :/
> 
> Contributions are welcomed.

Yep !
Once I find out how to do this and get to know more about the current options in ATAT, which I of course would not want to get in conflict with ....

mschnell1, 2023-05-11:
Sorry for continuing being a PITA. 
Trying to implement `AT+QMTPUBEX` , I see that ATAT successfully detects a "prompt" (`>`): In the log I see: 
`
DEBUG atat::ingress               > Received prompt (4/4)
`
Seemingly this does release the `client.send().awai`t; with an OK result. But how to see that this in fact is a pronpt and not an "OK" from the modem  ?

And after this I need to send a  certain count of bytes as the raw payload. 

How to do this ? I get close to tzhat by 
```

#[derive(Clone, Debug, AtatCmd)]
#[at_cmd(
    "",
    NoResponseMqttSetup,
    cmd_prefix = "",
    termination = "",
    value_sep = false,
    timeout_ms = 3000
)]
pub struct MqttSend2_1 {
    pub msg: Bytes<64>,
}

``` When trying to use a string type, the serializer creates additional quotes. and with other types such as Bytes, I can't seem to be able to define the count of bytes to be sent. 

-Michael

mschnell1, 2023-05-16:
@ijager : How do you progress with MQTT ? 
-Michael

ijager, 2023-05-19:
> @ijager : How do you progress with MQTT ?

I just implemented `+QMTPUBEX` by splitting it up in two parts. Seems to work as expected.

mschnell1, 2023-05-22:
Yep. That does work fine for me, as well. I just did not find out how to be sure that the callback I get between the parts in fact is a "Prompt" (which ATAT seems to be detecting) or something else (such as an `OK` or `Error` ), which would indicate a problem. 
Receiving ASCII (and supposedly UTF-2) data also works perfectly.

But In fact I would like to transfer binary data. The modem does handle this decently, but ATAT can't cope with a double quote or any bytes > 127 in the payload.

Thanks,
-Michael

ckrenslehner, 2025-03-31:
> I noticed the following when trying out `atat`. When defining ATI response using `heapless::Bytes` it parses correctly.
> 
> #[derive(Debug, Clone, AtatResp)]
> pub struct ATIResponse
> {
>     pub info: Bytes<64>,
> }
> 
> The response:
> 
> ATIResponse { info: b'Quectel\r\nEC25\r\nRevision: EC25EFAR06A09M4G' }
> 
> But when using `heapless::String` it results in `Error::Parse`.
> 
> #[derive(Debug, Clone, AtatResp)]
> pub struct ATIResponse
> {
>     pub info: String<64>,
> }
> 
> Perhaps still the same issue as in [#86](https://github.com/FactbirdHQ/atat/issues/86).

I also tried parsing the ATI response and came up with a custom parsing function using the `nom` crate. `atat` re-exports `nom` so it can be used straight away. 
IMO it is understandable, that the ATI response is not parsable in a straight-forward way, because the response parameters are separated by `\r\n` which is odd.

``` rust
const MAX_MANUFACTURER_LENGTH: usize = 10;
const MAX_MODEL_LENGTH: usize = 10;
const MAX_REVISION_LENGTH: usize = 20;

#[derive(Debug, Clone, AtatResp, PartialEq)]
#[cfg_attr(feature = "defmt", derive(defmt::Format))]
pub struct ModemInformation {
    pub manufacturer: String<MAX_MANUFACTURER_LENGTH>,
    pub model: String<MAX_MODEL_LENGTH>,
    pub revision: String<MAX_REVISION_LENGTH>,
}

impl ModemInformation {
    pub(crate) fn parse(bytes: &[u8]) -> Result<Self, atat::Error> {
        use core::str::FromStr;

        use atat::nom;
        use nom::{
            IResult,
            bytes::complete::{tag, take_till, take_until},
            character::complete::line_ending,
        };

        fn parse_inner(input: &str) -> IResult<&str, (&str, &str, &str)> {
            let (input, vendor) = take_until("\r\n")(input)?;
            let (input, _) = line_ending(input)?;

            let (input, model) = take_until("\r\n")(input)?;
            let (input, _) = line_ending(input)?;

            let (input, _) = tag("Revision: ")(input)?;
            let (input, revision) = take_till(|c| c == '\r' || c == '\n')(input)?;

            Ok((input, (vendor, model, revision)))
        }

        let (vendor, model, revision) =
            parse_inner(core::str::from_utf8(bytes).map_err(|_| atat::Error::Parse)?)
                .map_err(|_| atat::Error::Parse)?
                .1;

        Ok(ModemInformation {
            manufacturer: String::from_str(vendor).map_err(|_| atat::Error::Parse)?,
            model: String::from_str(model).map_err(|_| atat::Error::Parse)?,
            revision: String::from_str(revision).map_err(|_| atat::Error::Parse)?,
        })
    }
}
```

The response parsing function can be used in the macro via the `parse` attribute

``` rust
#[derive(Debug, Clone, AtatCmd)]
#[at_cmd("ATI", responses::ModemInformation, timeout_ms = 300, cmd_prefix = "", parse = responses::ModemInformation::parse)]
pub struct GetModemInformation;
```

This way parsing the ATI response into separate parameters works fine

``` rust
#[cfg(test)]
mod tests {
    #[test]
    fn read_ati() {
        let cmd = super::GetModemInformation;
        let response = cmd
            .parse(Ok(
                b"Quectel\r\nBG95M2\r\nRevision: BG95M2LAR01A01\r\n\r\nOK\r\n",
            ))
            .unwrap();
        log::info!("Response: {:?}", response);

        let exp = super::responses::ModemInformation {
            manufacturer: String::from_str("Quectel").unwrap(),
            model: String::from_str("BG95M2").unwrap(),
            revision: String::from_str("BG95M2LAR01A01").unwrap(),
        };

        assert_eq!(response, exp);
    }
}
```

mschnell1, 2025-04-04:
@MathiasKoch 
We now finally are staring to continue the project, as now there are Quectel modems with much lower average power consumption with only sporadic communication. this is what we  need with our battery driven IOT-type application. 
That is why I will restart with testing and finally will work on a "real" embedded platform (instead just prototype-testing on PC). For this I will need to combine ATAT with Embassy. 

Right now I did not get my old project to compile. 
I'll be back after some more investigation....

MathiasKoch, 2025-04-04:
Sounds exciting 👍 
Always nice with users, but even more so with contributors 👍

mschnell1, 2025-04-11:
I moved my project to a new computer, now working in WSL rather than directly on Windows (much faster :) ). 

Here I found that the feature "async" is not supported any more. 

Removing same creates a clean cargo.toml file, but of course lots of errors when compiling my source such as: 

```
error[E0277]: the trait bound `Rx: embedded_io_async::Read` is not satisfied
   --> src/atat_tokio.rs:178:5
    |
178 |     ingress.read_from(&mut rx).await
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_io_async::Read` is not implemented for `Rx`
```

Before diving in the docs...

Is there some small thing I can modify or will I to create the testing project completely from scratch ?

ckrenslehner, 2025-04-11:
In the examples directory is an example using embassy. If you create a buffered UART, both tx and rx should implement the embedded-io-async traits

mschnell1, 2025-04-11:
The old working (with MQTT) example I did did use Embassy (sitting on Tokyo for testing in a Windows PC).

Now I want to do software or en embedded ARM Cortex M 33 CPU without OS, hence of course using Embassy "pure". 

I'm not sure  if I should directly go for the embedded Hardware or first try to connect the Modem to the PC (now Linux) and have it do MQTT (porting the old test code I did).

ckrenslehner, 2025-04-11:
You can also just use `tokio` and `atat`. There is also an example in the dir. 
See https://github.com/FactbirdHQ/atat/blob/master/examples/src/bin/std-tokio.rs

So you could try the modem using your PC

ckrenslehner, 2025-04-11:
You are welcome :-)

mschnell1, 2025-04-11:
OK, I'll try all this next week.
Is there anything on behalf of MQTT  in the create ? (I seem to remember that I once provided some testing code hat did work for me)

mschnell1, 2025-04-11:
BTW.: should we go on discussing here : https://github.com/FactbirdHQ/atat/issues/149
(serch for
```
Ahh, yeah that is a very common way of doing it in AT commands, but it is not currently supported in ATAT :/
Contributions are welcomed. 
```

MathiasKoch, 2025-04-11:
@mschnell1 If i remember correctly, you were looking to a way to parse a response containing `<payload len>,<payload>`. e.g. `5,"abcde"`?

In that case, have a look at https://github.com/FactbirdHQ/atat/blob/master/serde_at/src/de/length_delimited.rs

mschnell1, 2025-04-12:
Many thanks !
I'll do so ASAP.
(what I want to achieve us using the MQTT functionality of the modem with binary data. I'll tel you know, what I find....)

ckrenslehner, 2025-04-12:
You could take a look at https://gitlab.com/scrobotics/optical-makerspace/dark-sky-meter-fw

They are using mqtt with the BG95. Maybe this is helpful :-)

mschnell1, 2025-04-12:
Great !
I'll take a look.
Thanks a lot !

mschnell1, 2025-04-24:
> In the examples directory is an example using embassy. If you create a buffered UART, both tx and rx should implement the embedded-io-async traits

I am trying to get the embassy example running on the hardware I currently use (a "Discovery kit for IoT node with STM32U5 series"). 

I could so some modifications to be able to have the code compile and run and seemingly some signal is outputed towards the modem. 

But I (of course() don't get an answer (yet).  I do get a huge log showing Embassy "ring buffer" activity.

As described in the example I do get a 1 sec timeout with each status/action.

In fact I don't know how to see any answer by the modem and appropriate ATAT reaction / result using the example code. 

What should I expect to see when the modem doe4s answer ? 

Do I need to add more code to the example to activate receiving ? 

Do I need to define interrupt services for the UART in my code or are those already prepared by the ATAT / Embassy infrastructure ?  

In another example I found (e.g.) `static mut INGRESS: Option<atat::IngressManager> = None;` but IngressManager does not seem to exist any more. 

Thanks a for for any help to get me started ....

mschnell1, 2025-04-25:
@MathiasKoch 
When adding the feature` "defmt"` to `atat`, I get;  
```
2.147552 DEBUG Sending command: b"AT+CGMM\r\n"
```
But this is the only log output I see from `atat`.  but many from 
`embassy_stm32::usart::find_and_set_brr` and `embassy_hal_internal::atomic_ring_buffer::{impl#?=}::pop_buf`

mschnell1, 2025-04-25:
I verified that The modem does answer something sensible to the four requests in the example 

I don't see anything of ATAT handling the answers in the log, And the timeout of a second still is happening every time. 

Do I do something wrong ? 

How to see the incoming messages in ATAT ? (and/or in the Embassy HAL log ? )

ckrenslehner, 2025-04-25:
Try increasing the log level in your Config.toml

Normally there are also logs when data is received. 

You can also try to separate concerns and test your atat setup without a modem. There is an example which mocks incoming data with Tokio.

mschnell1, 2025-04-25:
Thanks ! 
I'll try to find out how top do this.
The problem is that I am not perfectly sure that the  hardware works as I expect and the output from the modem indeed reaches the UART.

mschnell1, 2025-04-29:
Seems the signal does reach the appropriate pin of the µC Chip. 
going on trying to find out how that data streqam can be detected / traced in the software.

mschnell1, 2025-04-30:
I was able to see the response of the modem in the debug log: 
```
0.291076 DEBUG Received echo or whitespace (8/16): b"AT+CGMI\r\r\nQuecte"
└─ atat::ingress::{impl#1}::advance::{async_fn#0} @ /home/mschnell/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/atat-0.24.0/src/ingress.rs:228 
```
Now let's see if I can find out if/how `ATAT ` scans/detects in somehow. (right now I still get a timeout with 
```
let res = client.send(&GetManufacturerId).await;
```

mschnell1, 2025-06-03:
@MathiasKoch 
As I feature an unrelated  problem that blocks me, I open a new issue ....