FactbirdHQ/atat #150: URC parser does not allow colon

mschnell1, 2023-05-03:
The Modem I use (Quectel EC25) after connectomg to the Network sends something like 
`
+QMTOPEN: 0,3
`
I know that / how I can implement my own parser for respnses in client.send(), but this response is delayed, and hence I seem to need to add the ` #[at_urc("+QMTOPEN")]` to the Urc enum. 
Unfoprtuinately I get 
`
ERROR atat::ingress               > Parsing URC FAILED: "+QMTOPEN: 0,3"
`
and hence no return from `subscriber.next_message().await;`  
We already did test, that the stadard parser does not like colons. 
Is this intentionnally ?
Is there a way to handle such Midem responses ? 
-Michael

MathiasKoch, 2023-05-04:
I am not quite sure what is wrong in your case, but all the URC's i am using this for are using colon, and it works fine?

https://github.com/BlackbirdHQ/ublox-cellular-rs/blob/master/ublox-cellular/src/command/mod.rs

It's a bit hard without seeing your enum

mschnell1, 2023-05-04:
I have: 

```
pub enum Urc {
    #[at_urc("+QMTRECV")]
    QmtrecvReceived(mqtt::on_receive::OnReceived),
    #[at_urc("+QMTOPEN")]
    QmtopenReceived(mqtt::on_receive::OnReceived),
    #[at_urc("+QISTAT")]
    QistateReceived(mqtt::on_receive::OnReceived),
}
```
I can reproduce the problem with just one.

mschnell1, 2023-05-04:
(Not in the office right now)
AH, the `OnReceived` struct is used to configure the Parser. That does make sense. 
As I did not find information on that, I just put some random elements in that struct <actually meant for MQTT responses, but right now I am testing other stuff>. 
So I need to construct a decent OnReceive struct for responses such as `+QMTOPEN: 0,3 `
I'll try to find more information on that ... (
Thanks

mschnell1, 2023-05-05:
Yep !
with constructing an appropriate OnReceived struct such as 
```

#[derive(Clone, AtatResp)]
pub struct OnReceivedQmtOpen {
    pub p1: u32,
    pub p2: u32,
}
```
it works.
Thanks for your help !

closed 2023-05-05 by mschnell1