FactbirdHQ/atat #153: Null Parser

mschnell1, 2023-05-10:
I understand that with "derived" communication structs by ATAT, any response from the Modem (direct or URC ) is handled by a parser. The configuration for that parser is derived from the  elements of the appropriate struct that is to receive the information. 
If the detected message does not fit to the type of the elements, a "pars" error occurs, and the user software is not provided with the content of the message (not even "raw".)
I found that e.g. with `ATI` (and the modem I use - I guess any new user will try that command first ), you can't "easily" provide a struct that results in something not firing a pars Error. This is understandable, as the message sent by the modem is a non quoted string. The parser seems to be "normally" able to handle "sensible" stuff like numbers and quoted strings. In fact also for debugging it would be helpful to have the user software receive the raw (text) from the modem.

With the help of @ijager, I was able to construct a struct for that purpose by using an element of type  `heapless_bytes::Bytes<64>`. Other than with the rather standard e.g. `Vec<u8, 300>` , here I get the raw text. (I am not deep enough into Derive macros to understand why this is the case.) But anyway, it would be nice if ATAT would provide such a "null-parsed" type out of the box, so that we don't need to find and pull yet another crate for this. 
Thanks for listening,,
-Michael

mschnell1, 2023-05-11:
BTW.: 
On top of this it might also be a nice idea to allow for configuring the parser to use alternate formats for dedicated Receive structs. e.g. multiple unquoted strings separated by `/r/n` .

mschnell1, 2023-05-16:
I found that the atat feature "bytes" allows for using atat::heapless_bytes::Bytes without explicitely polling the heapless_bytes

MathiasKoch, 2023-12-22:
If I understand you correctly, this can now be reduced to a lack of proper documentation around the `bytes` feature?

closed 2023-12-22 by MathiasKoch