FactbirdHQ/atat #154: URC with length an binary content

mschnell1, 2023-05-13:
Some Modems send URCs with a payload length specification followed by a stream of non transparentized bytes. 
AFAIU, the ATAT "URC" Parser can't handle this. 
More generally, I seem to understand that the ATAT derive macros provide handling of "streaming" (serializing / deserializing / parsing) according to the types of the elements of a struct. Here unquoted binary information can be denoted by the `heapless_bytes::Bytes` type. But with same the payload length is defined by the Bytes count (when sending) and can't be denoted by another field in the message (when receiving). 
To me it seems that for the case in question it would be good if it would be possible to specify an alternate parser (e.g. by an attribute macro in the appropriate Urc enum arm or in the OnReceive struct denoted by same). By that any kind of message could be handled. 
(I would be inclined to contribute to ATAT in order to implement such an enhancement. But I am relatively new to Rust and only have a faint impression on macro programming. So I would need some support on finding a starting point....)
-Michael

mschnell1, 2023-05-24:
I found that the modem I use, when doing TCP transfer rather than MQTT transfer,  provides a similar answer scheme with the `+QIRD `block. Other than with `+QMTRECV`   here the binary information is separated from the length by `CR`    `LF`    ( rather than by  ` , `  ` " `  .
Did anybody already use such TCP communication via Moem by means of ATAT ?

mschnell1, 2023-05-24:
I found that it's possible to link one's own parser to an arm of the `Urx` enum by not deriving the `OnReceived...`  struct from `AtatResp` . 
For a test I did this:
```

#[derive(Clone /* , AtatResp*/)]
pub struct OnReceivedTest {
    pub text: atat::heapless_bytes::Bytes<0x100>,
}

impl<'a> atat::serde_at::serde::Deserialize<'a> for OnReceivedTest {
    fn deserialize<D>(deserializer: D) -> Result<Self, D::Error>
    where
        D: atat::serde_at::serde::Deserializer<'a>,
    {
        let r = OnReceivedTest {
            text: atat::heapless_bytes::Bytes::<0x100>::new(),
        };
        Ok(r)
    }
}

```

Now I suppose I can do the `deserialize()` functiom to decode any data from the modem following the appropriate keyword denoted by the `Urc` enum.

mschnell1, 2023-05-31:
I suppse the appropriate serde documentation will be needed to be adhered.
-> https://serde.rs/custom-serialization.html

MathiasKoch, 2023-11-22:
I think for this it would be nice if we could provide a struct type that can be used for the field, with the general structure along https://docs.rs/tokio-util/latest/tokio_util/codec/length_delimited/

ijager, 2023-12-04:
See #186

closed 2023-12-07 by MathiasKoch