FactbirdHQ/atat #166: Support for commands with both quoted and unquoted strings

ijager, 2023-07-26:
I want to send the following command:

```
AT+QSSLCFG="ciphersuite",2,0xFFFF
```

Note that it has both a quoted string `"ciphersuite"` and an unquoted string `0xFFFF`, I believe this is currently not possible to serialize with atat as the `quote_escape_strings` option is set for the entire command.

So I would like to add this `quote_escape_strings` option for `ArgAttributes` as an optional arg.

```rust
pub quote_escape_strings: Option<bool>,
```
 

I think if the option is set on `ArgAttributes` it should override the `CmdAttributes`.

Any thoughts?

I will work on a PR

natevw, 2025-12-16:
does the work in https://github.com/FactbirdHQ/atat/pull/135 help at all with this?

MathiasKoch, 2026-02-21:
Hi.

V.250 section 5.4.2.2 defines string type parameters as enclosed in double quotes. It's not optional — quoting is what makes it a string parameter in AT command grammar.

Unquoted values are numeric/enum parameters by definition.

In this particular case, you should probably use the provided https://github.com/FactbirdHQ/atat/blob/master/serde_at/src/de/hex_str.rs

Closing as non-issue.

closed 2026-02-21 by MathiasKoch