FactbirdHQ/atat #210: Unable to parse URC as bytes

cacharle, 2024-08-28:
I am failing to make a response struct that will parse to my desired URC:

```rust
#[derive(AtatUrc)]
pub enum Urc {
    #[at_urc("+QIND")]
    OTAStatus(OTAStatus),
}

#[derive(AtatResp)]
pub struct OTAStatus {
    bytes: heapless_bytes::Bytes<256>,
}
```

Gives me some of the following errors when parsing:

```
[ERROR] - Parsing URC FAILED: b"+QIND: \"FOTA\",\"HTTPSTART\""
[ERROR] - Parsing URC FAILED: b"+QIND: \"FOTA\",\"DOWNLOADING\",91%"
[ERROR] - Parsing URC FAILED: b"+QIND: \"FOTA\",\"HTTPEND\",0"
```

I have also tried to properly parse the fields of the response like so:

```rust
#[derive(AtatResp)]
pub struct OTAStatus {
        // fota: heapless::String<4>, // "FOTA"
    event: heapless::String<16>, // HTTPSTART, DOWNLOADING, HTTPEND, START, UPDATING, END
    // arg: Option<heapless::String<16>>, // percentage for download and update, result code for httpend and end
    // status1: Option<u8>,
    // status2: Option<u8>,

}
```

cacharle, 2024-08-28:
It is able to parse it if I don't specify any struct in the URC enum:

```rust
#[derive(AtatUrc)]
pub enum Urc {
    #[at_urc("+QIND")]
    OTAStatus,
}
```

cacharle, 2024-08-29:
Okay, looking has some previous code that could parse it, I think this specific URC is terminated by a `\r` instead of `\r\n`.

cacharle, 2024-08-29:
Also tried to copy the default urc parser in my code and replace the `\r\n` by `\r` but it wasn't successful

ckrenslehner, 2025-04-01:
I can imagine, that there is some problem with the escaped quotes `\"`. Also, the position of the `,` seems odd?
As a fallback, you could try to parse it with a custom `parse` function using `nom`. Something similar to [this](https://github.com/FactbirdHQ/atat/issues/149#issuecomment-2765834970) sample

ckrenslehner, 2025-04-01:
Okay, my comment was not correct. Maybe you can make something like [this](https://github.com/FactbirdHQ/atat/blob/master/atat/src/ingress.rs#L330-L335) work.

cacharle, 2025-04-02:
I am now unable to reproduce this issue, thank you for your time nevertheless

closed 2025-04-02 by cacharle