FactbirdHQ/atat #211: `Ingress::read_from` causes busy-loop if write_buf is full

t-moe, 2024-09-30:
The current implementation of `Ingress::read_from` can cause the MCU to busy-loop: If for some reason the write_buf becomes full, the embedded_io_async::Read will return/error immediately, causing the MCU to busy-loop.

https://github.com/FactbirdHQ/atat/blob/2768e5ae760992641defc048972c2aa7e12043d2/atat/src/ingress.rs#L57-L76

As to why the write_buf became full in my case, I'm still investigating.

MathiasKoch, 2024-09-30:
Nice catch

t-moe, 2024-09-30:
Am I correct that ATAT should in theory be able to recover from garbage data at the beginning of the buffer?
If so, we could just print a warning and self.clear() if buf.len()==0

MathiasKoch, 2024-09-30:
I would say so yeah.

I have been meaning to set up a fuzz test on it, to verify this behaviour.

PR #212 (2024-09-30): Protect against busy-loop in Ingress::read_from
Fixes #211
closed 2024-09-30 by MathiasKoch