FactbirdHQ/atat #219: ESP URC IP assigment notification classified as `echo or space`.

marius-meissner, 2025-02-06:
While migrating [esp-at-nal](esp-at-nal) to the newest atat version (s. https://github.com/atlas-aero/rt-esp-at-nal/issues/19) I stumbled across some minor possible regression issue.
Seems the URC notice `WIFI GOT IP\r\n` gets classified as `echo or space`.

Interestingly, the first message `WIFI CONNECTED\r\n` is reaching the URC parser.

But I'm also not sure whether these messages or their format might violate the ATC URC specification?

This is the log:
```
[DEBUG] [atat::ingress]: Received echo or space (11/16): "AT+CWMODE=1\r\n\r\nO"
[DEBUG] [atat::ingress]: Received OK (8/8)

[DEBUG] [atat::ingress]: Received echo or space (52/54): "AT+CWJAP=\"****\",\"******\"\r\n"

[DEBUG] [atat::ingress]: Received URC/16 (18/18): "WIFI CONNECTED\r\n"
[DEBUG] [atat::ingress]: Received echo or space (11/13): "WIFI GOT IP\r\n"
[DEBUG] [atat::ingress]: Received OK (8/8)

[DEBUG] [atat::ingress]: Received echo or space (8/16): "AT+CIFSR\r\n+CIFSR"
[DEBUG] [atat::ingress]: Received response (72/72): "+CIFSR:STAIP,\"192.168.0.45\"\r\n+CIFSR:STAMAC,\"4c:eb:d6:ed:9a:22\""
```

Any thoughts from your side on this?

In any case, a big thank you for the continuous development of atat.

MathiasKoch, 2025-02-07:
Huh..

The interesting part here, I think, is that the `WIFI CONNECTED` is correctly caught..?

I am not really sure what happens here. We don't have any URC's in a clear-text format like these ourselves, but I think ATAT should support it. As this is not quite the case currently, a PR is more than welcomed!

marius-meissner, 2025-02-07:
Ok, after some debugging, the issue is not related to atat.
Normally, these blank URC messages are terminated with double CRLF.
But in the case of `WIFI CONNECTED` and `WIFI GOT IP` there is only a single CRLF in between.

Not sure if this is an exception in the spec or an inconsistency in the modem's firmware.
Anyway, this can be handled in the URC parse on my side.

closed 2025-02-07 by marius-meissner