FactbirdHQ/atat #221: Parsing `OK` followed by response

tshakah, 2025-03-05:
I'm trying to integrate the SIM7600 modem (https://files.waveshare.com/wiki/SIM7670G-LTE-Cat-1-GNSS-HAT/SIM767XX_Series_AT_Command_Manual_V1.01.pdf) using this crate, but some of the responses returned have `OK` first, then the data - e.g. `CMQTTSTART` (ref page 266 of that manual) returns:
```
OK
+CMQTTSTART: 0 | <errcode>
```

It looks like `atat` is ignoring the codes that are returned because it ignores everything after the `OK`. Is this correct? Is there a workaround? I tried implementing a custom parser via the `at_cmd` attribute but it is called after the OK so has an empty response.

MathiasKoch, 2025-03-05:
Hi!

That sounds like a bit of a cursed module 😄 I think @rmja has been working with SIMCOM, maybe he has some experience to share here?

Apart from that, it should absolutely be possible to support, but might require a bit of effort from your side. You were on the right track with the custom implementation, but rather that `at_cmd`, you need to implement a custom `Digester`.

In ATAT, the digester is the parser (default implementation powered by nom, but you can choose anything you want), and the `at_cmd` part is deserializing, powered by serde.

Digester: https://github.com/FactbirdHQ/atat/blob/master/atat/src/digest.rs

rmja, 2025-03-05:
Year, those modems are insane. I have created a sim800 and sim900 driver here which uses atat: https://github.com/rmja/chip-drivers/tree/master/cellular/simcom-gprs. It is reasonably featured, and has a lot of custom parsing to handle the modem peculiarities.

tshakah, 2025-03-06:
Thanks, I suspected I'd have to go deeper than just the parser. I've looked into a custom digester and I am not sure that is enough - there is enough of a delay between the `\r\nOK\r\n` and the rest of the response that the `custom_success` parser isn't called. There might be ways I can overcome this, but they are all a little convoluted. I think I might go down treating all the reversed responses as URCs and then use them to update a state machine that way.

Sadly I have to use this module in this project, but for future work I'll have more say over which ones to use so will be looking at alternatives.

rmja, 2025-03-06:
I am doing something similar,  for example here https://github.com/rmja/chip-drivers/blob/master/cellular/simcom-gprs/src/services/data/tcp.rs#L81

The issue with simcom modems is that you cannot reliably assume that it is always after ok. If the response is listed in the urc section in the docs, then it can arrive both before or after the ok. I have seen this for many of the commands.

tshakah, 2025-03-07:
Thanks for the responses. As this isn't an issue with `atat` I'll close this, and hopefully be able to publish code with some measure of control over the SIM7600 at some point in the future

closed 2025-03-07 by tshakah